dektronics/printalyzer-densitometer #4: Provide a way for end-users to update device firmware

dkonigsberg, 2021-07-26:
While it is the hope that a "launch" version of the device will effectively have finalized firmware, the possibility for future updates to fix bugs and add features will always remain.

Its probably not reasonable to expect average end-users to use an ST-Link with a Tag-Connect cable to install firmware updates on the device. Therefore, another option should be provided.

The biggest issue right now is that the current firmware build (which is not feature-complete, but is close) already uses up most of the available flash on the STM32L052 microcontroller. Furthermore, the built-in bootloader options on this MCU are really not suitable for handling end-user firmware updates.

The solutions to this problem can be broken up into two categories:
- Optimize the build enough that we can make 8-16kb of space, and use a 3rd party bootloader to facilitate updates
- Switch to a 128kb+ MCU that has a built-in USB DFU bootloader and provide a means of using that
  - Right now, due to the great microcontroller shortage, options for this are limited and expensive (especially within the L0 line).

Options for a tiny cram-able bootloader include these projects:
- [LED Wristwatch Bootloader](https://github.com/kcuzner/led-watch/tree/master/bootloader)
  - Actually designed for the STM32L052
  - Fits in 8kb
  - Works by simulating a USB HID device, so some sort of custom software will be required on the host side to facilitate installing firmware updates
  - MIT License
- [DFU Bootloader](https://github.com/IntergatedCircuits/DfuBootloader)
  - Also claims to fit within 8kb
  - Works via USB DFU, which is preferred
  - Does not support the L0, so it will require porting.  (Does support the F0, F3, F4, and L4. No idea how similar they are.)
  - Apache License 2.0

dkonigsberg, 2021-10-10:
As of recent updates, there is now a very basic way of doing this:
1. Hold down boot selector button with device powered off
2. Plug in device to a computer
3. Device shows up as a USB DFU device (with both LED sets on, for some reason)
4. `$ dfu-util -a 0 --dfuse-address 0x08000000 -D densitometer.bin`
5. Power-cycle the device

A more polished version is going to need a few additions, that may or may not end up as separate issues:
- Add instructions and/or build steps to run `dfuse-pack.py` to make a proper DFU file
- Implement version descriptor data into the firmware build
- Add a way to query this version descriptor, both via the firmware itself and via a scripted sequence involving `dfu-util` and some processing of its output.
- Add a firmware feature to make the device restart itself into DFU mode without needing a hidden BOOT0 button and power cycle (but keep those for cases where the firmware has malfunctioned).
- Wrap this all up into an updater utility, either as a script or as part of a future desktop application.

dkonigsberg, 2021-11-11:
There are now two ways to update on-device firmware, so I think its time to close this issue:

1. The DFU method, as mentioned above. This is great for programming from a bare-metal state, and/or fixing a bricked device. It might be a little too involved for ordinary updates.
2. An embedded UF2 bootloader, triggered by holding down a button combination on power-up. This is very easy for an ordinary end-user to use to install updates. It may still have room for improvement, but in its current state it gets the job done.

closed 2021-11-11 by dkonigsberg

pixelwaster, 2021-11-11:
You might want to check out the nanoVNA, The update process is very easy and they use a similar processor. Basically select the file, select the device, and update.

Very interesting project. Looking forward to seeing how this goes almost more than the densitometer itself.

dkonigsberg, 2021-11-11:
> You might want to check out the nanoVNA, The update process is very easy and they use a similar processor. Basically select the file, select the device, and update.
NanoVNA uses DFU. There are a few utilities that can be used to do a DFU update, some easier to get going than others, but none really intended for normal end users.  A polished DFU update process would require writing my own wrapper, and then dealing with OS-specific device issues.

The advantage of UF2 is that it requires no special software.  Just boot into UF2 mode, the device shows up as a USB mass storage device, drag-and-drop the file to it, and you're done.